YoWASP/yosys #1: Non-zero error code causes nmigen to fail

DavidPowell, 2020-07-01:
Running the nmigen example `01_blinky.py` with nmigen 0.3.dev95+g126f0be and yowasp-yosys 0.9.post4547.dev5 on Windows 10 x64 yields the following error messages:

```
Traceback (most recent call last):                                                                                           
  File "01_blinky.py", line 21, in <module>                                                                                  
    platform.build(Blinky(), do_program=True)                                                                                
  File "<my path>\nmigen\nmigen\build\plat.py", line 90, in build                         
    plan = self.prepare(elaboratable, name, **kwargs)                                                                        
  File "<my path>\nmigen\build\plat.py", line 163, in prepare                      
    return self.toolchain_prepare(fragment, name, **kwargs)                                                                  
  File "<my_path>\nmigen\nmigen\build\plat.py", line 437, in toolchain_prepare            
    render(content_tpl, origin=content_tpl))                                                                                 
  File "<my_path>\nmigen\nmigen\build\plat.py", line 431, in render                       
    "autogenerated": autogenerated,                                                                                          
  File "<my_path>\Miniconda3\lib\site-packages\jinja2\environment.py", line 1090, in render                            
    self.environment.handle_exception()                                                                                      
  File "<my_path>\Miniconda3\lib\site-packages\jinja2\environment.py", line 832, in handle_exception                   
    reraise(*rewrite_traceback_stack(source=source))                                                                         
  File "<my_path>\Miniconda3\lib\site-packages\jinja2\_compat.py", line 28, in reraise                                 
    raise value.with_traceback(tb)                                                                                           
  File "<template>", line 2, in top-level template code                                                                      
  File "<my_path>\nmigen\nmigen\build\plat.py", line 315, in emit_debug_verilog           
    strip_internal_attrs=False, write_verilog_opts=opts)                                                                     
  File "<my_path>\nmigen\nmigen\back\verilog.py", line 10, in _convert_rtlil_text         
    yosys = find_yosys(lambda ver: ver >= (0, 9))                                                                            
  File "<my_path>\nmigen\nmigen\_yosys.py", line 220, in find_yosys                       
    version = proxy.version()                                                                                                
  File "<my_path>\nmigen\nmigen\_yosys.py", line 157, in version                          
    version = cls.run(["-V"])                                                                                                
  File "<my_path>\nmigen\nmigen\_yosys.py", line 186, in run                              
    return cls._process_result(popen.returncode, stdout, stderr, ignore_warnings, src_loc_at)                                
  File "<my_path>\nmigen\nmigen\_yosys.py", line 108, in _process_result                  
    raise YosysError(stderr.strip())                                                                                         
nmigen._yosys.YosysError
```

Running with `-m pdb` shows that `popen.returncode` being non-zero is the cause of this error. This can be verified independently by running `yowasp-yosys -V` then `echo %errorlevel%` which prints:
-1073741784

whitequark, 2020-07-01:
That return code corresponds to the `NTSTATUS` code `STATUS_BAD_STACK`, which is definitely concerning. Does `yowasp-yosys -V` actually print the version?

whitequark, 2020-07-01:
Nevermind, I can reproduce it.

whitequark, 2020-07-01:
Upstream issue: https://github.com/bytecodealliance/wasmtime-py/issues/36

whitequark, 2020-07-05:
New upstream issue: https://github.com/bytecodealliance/wasmtime/issues/1967

whitequark, 2020-07-08:
Fixed upstream.

whitequark, 2020-07-08:
You can upgrade wasmtime with `pip install -U wasmtime`. I'll bump the dependency shortly.

closed 2020-07-08 by whitequark

DavidPowell, 2020-07-09:
Thanks. I can confirm that the fix worked after I manually updated wasmtime.