YoWASP/yosys #17: YoWASP yosys fails to generate SVG diagrams

rw1nkler, 2020-09-14:
Currently, YoWASP Yosys fails to generate SVG diagrams. However, the `.dot` file is created correctly in the output directory. 
The problem seems to be related to `dot` execution:
```
3. Generating Graphviz representation of design.
Writing dot description to `yosys.dot'.
Dumping module ADDER to page 1.
Exec: dot -Tsvg 'yosys.dot' > 'yosys.svg.new' && mv 'yosys.svg.new' 'yosys.svg'
```

Steps to reproduce:
```
wget https://raw.githubusercontent.com/SymbiFlow/sphinxcontrib-verilog-diagrams/master/tests/verilog/adder.v
python3 -c "from yowasp_yosys import *; run_yosys(['-p', 'prep -top ADDER ; cd ADDER; ; show -format svg -prefix yosys', 'adder.v'])"
```

Comparison with standard Yosys:
```
wget https://raw.githubusercontent.com/SymbiFlow/sphinxcontrib-verilog-diagrams/master/tests/verilog/adder.v
yosys -p 'prep -top ADDER ; cd ADDER; ; show -format svg -prefix yosys' adder.v
```

whitequark, 2020-09-15:
Yeah, there needs to be a custom hostcall used to implement Yosys' `exec_command`. I had planned to implement it but other tasks were higher priority.

whitequark, 2024-10-04:
In the end I feel that this functionality is not important enough to warrant being implemented. Plus it would cause difficulty for the JavaScript port.

closed 2024-10-04 by whitequark