YoWASP/yosys #20: TCL support for yowasp-yosys

kammoh, 2020-12-14:
Would it be possible to add TCL support to `yowasp-yosys`?

whitequark, 2020-12-14:
Quite possibly; I would have to look closer at build requirements of Tcl to answer for sure.

kammoh, 2020-12-14:
That would be really great! Thanks!

whitequark, 2020-12-20:
I've tried to build Tcl for WASI. Unfortunately, in spite of its reputation for portability, Tcl has a large amount of dependencies on Unix-specific APIs (especially socket ones are problematic, though there are many more than that), and getting it to build and run under WASI is a significant undertaking.

I will not be porting Tcl to WASI, but if someone else does this, I will consider shipping a third-party port as a part of YoWASP.

closed 2020-12-20 by whitequark