YoWASP/yosys #21: Browser package (npm)

nturley, 2021-03-13:
This was discussed on twitter a few times but I thought it would be helpful to track it here.

## Use Cases
* netlist visualization
  * type verilog in the browser and see the netlist update in real time
* resource usage visualization
  * I started working on this thing that would fetch fusesoc projects from github, synthesize them and display resource usage as a sunburst diagram
  * Running it client-side prevents users from using it as synthesis-as-a-service, we are using the users CPU cycles, not my CPU cycles
* Place and route visualization
  * I havent tried this but it would be interesting to play with 
* Simulation with cxxrtl
  * Useful for FPGA workshops where you dont want to deal with machine setup or user hardware

## Why
 * As a quick demo of Yosys. Some hardware engineers are not very computer savvy and explaining how to install python and running pip isn't as easy as sending them a link to a webpage. Seeing browser visualizations also makes for an impressive flashy demonstration. Also simply being able to run synthesis in browser demonstrates how light weight yoys is compared to vendor tools.
 * Introductory digital logic workshops are much easier when there is literally zero computer setup required
 * A lot of visualization tools are already in JS (netlistsvg, wavedrom, d3, etc)
 * People keep on doing one-off browser builds of Yosys for their specific use case. It would be nice if there was a good package that was good enough for everyone to use instead of rolling their own.

## What's the next best alternative?
 * Grab one of the builds of YosysJS floating around.
 * Jupyter notebooks are really great educational tool, but require learning the jupyter interface and at least a little python

## Challenges

Accessing files is annoying in the browser. In addition to the wasm file, you also need to deal with the users files, the tech library files. So, ideally those would be packaged in a way that is easy to use.

whitequark, 2021-03-13:
This is possible and in scope. I have a prototype already and will look into publishing it as a complete work later.

whitequark, 2021-03-13:
> Place and route visualization

Note that Qt nextpnr builds are out of scope at the moment. That's an enormous and highly complex dependency that I am not ready to support.

> Simulation with cxxrtl

You'd have to also use clang.wasm targeting wasm. That's possible although the simulation speed might not be so great.

nturley-copia, 2021-03-13:
If there's anything I can do to help, I'm happy to contribute.

whitequark, 2021-03-13:
Not really, the limiting factor is my time.

whitequark, 2023-12-08:
@nturley https://github.com/YoWASP/yosys/pull/28

PR #28 (2023-12-10): Add JavaScript package
Example:

```js
import { runYosys } from '../gen/bundle-node.js';

const files = await runYosys(
  ["inv.v", "-p", "hierarchy -auto-top", "-p", "synth_ice40 -json top.json"], 
  {"inv.v": "module inv(input a, output o); assign o = ~a; endmodule"});
console.log(files["top.json"]);
```

Fixes #21
closed 2023-12-10 by whitequark