YoWASP/yosys #22: assert failure on modern intel cpus with avx512_bf16

programmerjake, 2021-07-02:
running into this issue: https://github.com/gz/rust-cpuid/issues/45#issuecomment-873001709

you'll probably need to update to a later version of wasmtime once the fix is included in the version released on PyPI.

whitequark, 2021-07-02:
Got a link to a wasmtime issue?

programmerjake, 2021-07-02:
not yet, but here's the irc log:
https://libre-soc.org/irclog/latest.log.html#t2021-07-02T10:43:34

programmerjake, 2021-07-02:
wasmtime bug: https://github.com/bytecodealliance/wasmtime/issues/3053

whitequark, 2021-07-02:
Thank you for filing!

programmerjake, 2021-07-03:
Apparently this can be fixed just by updating the wasmtime dependency to v0.23 or later which no longer depends on raw-cpuid:
https://github.com/bytecodealliance/wasmtime/issues/3053#issuecomment-873045298

whitequark, 2021-07-03:
Fixed in 505780d.

whitequark, 2021-07-03:
Released on PyPI.

closed 2021-07-03 by whitequark