YoWASP/yosys #23: Support for Mac M1

hossein1387, 2021-09-29:
I am able to install `yowasp` using: `python -m pip install yowasp-yosys` but when I try to run it I get:

`RuntimeError: unsupported architecture for wasmtime: arm64`

Any plan to support arm64 for Mac M1?

whitequark, 2021-09-29:
This would require ARM64 Mac support in the backend we use, [wasmtime](https://github.com/bytecodealliance/wasmtime). Currently it only supports ARM64 Linux, but feel free to open an issue upstream!

hossein1387, 2021-09-29:
will do, thanks!

hossein1387, 2021-09-29:
Following the link you sent, I found that most of the works is already done: https://github.com/bytecodealliance/wasmtime-go/issues/53. However, for stable a release, they need a CI and Github actions currently does not support [VMs for Mac M1](https://github.com/actions/virtual-environments/issues/2187).

closed 2021-09-29 by hossein1387