YoWASP/yosys #24: Error when used with VSCode and Teroshdl

BharathS11, 2021-10-05:
Hi, When I use yowasp to generate netlist with Teroshdl and vscode I get the following error. Please take a look at the attached file. Thank you
[netlist_error.log](https://github.com/YoWASP/yosys/files/7284781/netlist_error.log)

whitequark, 2021-10-05:
Okay, there are several separate issues here:
  * First, absolute paths currently don't work on Windows. I haven't used TerosHDL but check if you can get it to pass relative paths to YoWASP. (This is still a bug in YoWASP, just one that might not be trivial to address.)
  * Second, there's infinite recursion in `shutil.rmtree` (called from `tempfile.TemporaryDirectory` destructor) that happens only when the first issue is triggered. I'm not sure why.

whitequark, 2021-10-31:
@BharathS11 Can you please confirm whether you were able to get the updated package to work?

whitequark, 2023-02-06:
> ` yowasp-yosys -p "read_verilog -sv ..\..\..\Documents\Projects\test\frontend\dsp\rtl\dsp_xcc_v2.v; proc; opt; write_json C:\Users\bhara\.teroshdl_yosys_output.json; stat"`

Both of these paths should work since commit https://github.com/YoWASP/yosys/commit/4ae03d98ee8175a65e9eb7eb736aa15120db9457, closing. Please comment if there are still any issues.

closed 2023-02-06 by whitequark

qarlosalberto, 2024-05-20:
I'm improving the support for `yowasp-yosys` in TerosHDL. Currently, is it possible to use absolute paths in Windows and Linux? Thanks.

qarlosalberto, 2024-05-20:
Great!!

I'm doing some tests. This command works fine:

```
yowasp-yosys -p 'read_verilog -sv /home/carlos/repo/vscode-terosHDL/packages/colibri/tests/yosys/helpers/verilog/mylib/counter_logic.v; read_verilog -sv /home/carlos/repo/vscode-terosHDL/packages/colibri/tests/yosys/helpers/verilog/counter_top.v; ; hierarchy -top counter_top; proc; ; write_json /home/carlos/hola.json; stat'
```

But not this command:

```
yowasp-yosys -p 'read_verilog -sv /home/carlos/repo/vscode-terosHDL/packages/colibri/tests/yosys/helpers/verilog/mylib/counter_logic.v; read_verilog -sv /home/carlos/repo/vscode-terosHDL/packages/colibri/tests/yosys/helpers/verilog/counter_top.v; ; hierarchy -top counter_top; proc; ; write_json /tmp/hola.json; stat'
```

There is no errors. But in the second command `yowasp-yosys` doesn't create the file `/tmp/hola.json`. I am doing something wrong?

whitequark, 2024-05-20:
The `/tmp` subtree is internally mounted to a temporary directory created just for this purpose, as it is unconditionally used by Yosys for e.g. running abc. The `/tmp` directory is [hardcoded in wasi-libc sources](https://github.com/WebAssembly/wasi-libc/blob/a3ef1520ec5c00f375eb87903151307c90cca4f1/libc-top-half/musl/src/stdio/tmpnam.c#L15) so you'll [need to pick a different temporary location](https://github.com/YoWASP/runtime-py/blob/c26c8d4dc8c0b1491ce06a26aac36d2ac6489f21/yowasp_runtime/__init__.py#L80-L89).

The same is true of `/share`, for different reasons.

qarlosalberto, 2024-05-20:
Ahh okok, thanks!! I will choose other path.