YoWASP/yosys #26: out of bounds memory access

karlp, 2022-06-24:
This is potentially user error, or a wasmtime error, I'm not experienced in this field, but I was attempting to build [orbtrace-mini](https://github.com/orbcode/orbtrace) using the ```yowasp-yosys``` and ```yowasp-nextpnr-ecp5``` instead of getting the binary oss-cad-suite tooling.

packages installed in the venv, via pip.
```
wasmtime==0.30.0
yowasp-nextpnr-ecp5==0.3.post31.dev324
yowasp-yosys==0.17.post67.dev359
```

The build works fine for quite a while, but eventually crashes with ```wasmtime._trap.Trap: wasm trap: out of bounds memory access```

```
.... much skipped ...
10.43.1.1. Executing ABC.
Running ABC command: /yosys-abc -s -f <abc-temp-dir>/abc.script 2>&1
ABC command line: "source /tmp/yosys-abc-000000/abc.script".

+ read_blif /tmp/yosys-abc-000000/input.blif 
+ read_lut /tmp/yosys-abc-000000/lutdefs.txt 
+ strash 
+ ifraig 
+ scorr 
Warning: The network is combinational (run "fraig" or "fraig_sweep").
+ dc2 
+ dretime 
+ strash 
+ dch -f 
+ if 
+ mfs2 
+ dress 
Traceback (most recent call last):
  File "/home/karlp/src/orbtrace.git/.env3/bin/yosys", line 8, in <module>
    sys.exit(_run_yosys_argv())
  File "/home/karlp/src/orbtrace.git/.env3/lib64/python3.9/site-packages/yowasp_yosys/__init__.py", line 79, in _run_yosys_argv
    sys.exit(run_yosys(sys.argv[1:]))
  File "/home/karlp/src/orbtrace.git/.env3/lib64/python3.9/site-packages/yowasp_yosys/__init__.py", line 75, in run_yosys
    return _run_wasm_app("yosys.wasm", ["yowasp-yosys", *argv])
  File "/home/karlp/src/orbtrace.git/.env3/lib64/python3.9/site-packages/yowasp_yosys/__init__.py", line 68, in _run_wasm_app
    app.exports(store)["_start"](store)
  File "/home/karlp/src/orbtrace.git/.env3/lib64/python3.9/site-packages/wasmtime/_func.py", line 100, in __call__
    raise WasmtimeError._from_ptr(error)
  File "/usr/lib64/python3.9/contextlib.py", line 126, in __exit__
    next(self.gen)
  File "/home/karlp/src/orbtrace.git/.env3/lib64/python3.9/site-packages/wasmtime/_func.py", line 263, in enter_wasm
    raise trap_obj
wasmtime._trap.Trap: wasm trap: out of bounds memory access
wasm backtrace:
    0: 0xaa2465 - <unknown>!<wasm function 16189>
    1: 0x1ad149 - <unknown>!<wasm function 2389>
    2: 0x1ace11 - <unknown>!<wasm function 2388>
    3: 0x1addc8 - <unknown>!<wasm function 2394>
    4: 0x1aebdd - <unknown>!<wasm function 2401>
    5: 0x1b167b - <unknown>!<wasm function 2433>
    6: 0x1b6630 - <unknown>!<wasm function 2501>
    7: 0x1b66a9 - <unknown>!<wasm function 2506>
    8: 0x5a3fcd - <unknown>!<wasm function 7481>
    9: 0x5a442a - <unknown>!<wasm function 7482>
   10: 0x5a5eca - <unknown>!<wasm function 7484>
   11: 0x5a69c3 - <unknown>!<wasm function 7487>
   12: 0x105297 - <unknown>!<wasm function 1414>
   13: 0x105445 - <unknown>!<wasm function 1415>
   14: 0x105a54 - <unknown>!<wasm function 1419>
   15: 0x146c3f - <unknown>!<wasm function 1754>
   16: 0x26289 - <unknown>!<wasm function 270>
   17: 0x27bf1 - <unknown>!<wasm function 289>
   18: 0xa6155 - <unknown>!<wasm function 1016>
   19: 0x26289 - <unknown>!<wasm function 270>
   20: 0x27bf1 - <unknown>!<wasm function 289>
   21: 0x1ba289 - <unknown>!<wasm function 2602>
   22: 0xd5bd7f - <unknown>!<wasm function 19128>
   23: 0xd519f8 - <unknown>!<wasm function 19126>
   24: 0x91643d - <unknown>!<wasm function 13541>
   25: 0x9160b1 - <unknown>!<wasm function 13539>
   26: 0x91993b - <unknown>!<wasm function 13579>
   27: 0x1224c94 - <unknown>!<wasm function 24315>
   28: 0x919aad - <unknown>!<wasm function 13581>
   29: 0x12230cc - <unknown>!<wasm function 24313>
   30: 0x91643d - <unknown>!<wasm function 13541>
   31: 0x9160b1 - <unknown>!<wasm function 13539>
   32: 0x8fce3d - <unknown>!<wasm function 13296>
   33: 0x8f93fa - <unknown>!<wasm function 13275>
   34: 0x8e4216 - <unknown>!<wasm function 13026>
   35: 0x8e4292 - <unknown>!<wasm function 13027>
   36: 0x8e352f - <unknown>!<wasm function 13003>
   37: 0xb773 - <unknown>!<wasm function 24>

Traceback (most recent call last):
  File "/home/karlp/src/orbtrace.git/./orbtrace_builder.py", line 112, in <module>
    main()
  File "/home/karlp/src/orbtrace.git/./orbtrace_builder.py", line 101, in main
    builder.build(**trellis_argdict(args), run=args.build)
  File "/home/karlp/src/orbtrace.git/deps/litex/litex/soc/integration/builder.py", line 350, in build
    vns = self.soc.build(build_dir=self.gateware_dir, **kwargs)
  File "/home/karlp/src/orbtrace.git/deps/litex/litex/soc/integration/soc.py", line 1193, in build
    return self.platform.build(self, *args, **kwargs)
  File "/home/karlp/src/orbtrace.git/deps/litex/litex/build/lattice/platform.py", line 43, in build
    return self.toolchain.build(self, *args, **kwargs)
  File "/home/karlp/src/orbtrace.git/deps/litex/litex/build/lattice/trellis.py", line 236, in build
    _run_script(script)
  File "/home/karlp/src/orbtrace.git/deps/litex/litex/build/lattice/trellis.py", line 174, in _run_script
    raise OSError("Error occured during Yosys/Nextpnr's script execution.")
OSError: Error occured during Yosys/Nextpnr's script execution.
```

System is fedora34, x86-64.

whitequark, 2022-06-26:
This is almost certainly an upstream bug in abc.

karlp, 2022-06-27:
Good enough for me, thanks.  It build fine with the oss-cad-suite binary packages, but almost definitely different versions of tooling there too.

closed 2022-06-27 by karlp

whitequark, 2022-06-27:
> It build fine with the oss-cad-suite binary packages

Try running the oss-cad-suite packages under Valgrind or AddressSanitizer; there is a chance you'll see the same error. Wasm happens to be more strict in what it allows than native code.

hstarmans, 2022-08-20:
This issue should be reopened.. Yowasp-yosys doesn't work due to it. I tested on windows and linux arm64.

whitequark, 2022-08-20:
Please post complete steps to reproduce in a new issue.

hstarmans, 2022-08-23:
Thanks for your effort on this.
Here is a repository with the files, steps to reproduce it and error.
https://github.com/hstarmans/bugreport