YoWASP/yosys #33: ESM Module problem

qarlosalberto, 2024-04-15:
I'm trying to use `@yowasp/yosys` npm package directly in a VSCode extension, but I'm encountering some difficulties. `@yowasp/yosys` uses `"type": "module",`, but VSCode haven't a good support for ES Modules. 

I'm having problems importing `@yowasp/yosys` with my commonjs extension. Also I have tried:

```
async function getYosys(): Promise<any> {
    const yosys = await (eval(`import('@yowasp/yosys')`) as Promise<typeof import('@yowasp/yosys')>);
    return yosys;
}

export async function showYosysVersion() {
    const yosys = await getYosys();
    await yosys.runYosys(["--version"]);
}
```

But there are some errors related to WebAssembly. Could be possible to publish the package with commonjs instead of ESM Module?

Thanks!

whitequark, 2024-04-15:
VS Code 1.88 should have been released recently with a recent enough Node.js to have some ESM support, but I haven't checked that yet. Is this with the latest VS Code?

qarlosalberto, 2024-04-15:
Yes, but I see that the problem when I use `eval` is the same that: https://github.com/YoWASP/yosys/issues/32

I think that you can close the issue.

closed 2024-04-15 by whitequark