YoWASP/yosys #35: frontend error

qarlosalberto, 2024-06-20:
One TerosHDL user is having some problems with `yowasp-yosys`, but I can't find the error. Have you any idea about what could be the error? He is using Windows.

[image]

whitequark, 2024-06-20:
That is not specific to `yowasp-yosys`. This happens if you pass a literal `;` as an argument to Yosys:

```
$ ./yosys ';'

 /----------------------------------------------------------------------------\
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |  Copyright (C) 2012 - 2024  Claire Xenia Wolf <claire@yosyshq.com>         |
 |  Distributed under an ISC-like license, type "license" to see terms        |
 \----------------------------------------------------------------------------/
 Yosys 0.39+1 (git sha1 f7153573c, ccache clang++ 14.0.6 -O0 -fPIC)
ERROR: Can't guess frontend for input file `;' (missing -f option)!
```

So, the issue is somewhere in your command line handling code. I don't know where exactly but calling cmd.exe twice looks suspect.

closed 2024-06-20 by whitequark

qarlosalberto, 2024-06-21:
Thanks! I have a good starting point.