YoWASP/yosys #37: Stack overflow on moderately deep ternary expression

rudy-6-4, 2024-08-22:
When generating code deep ternary, a stack overflow happens for moderate sequence of ternary.
Is there a way to have bigger/unlimited stacks for yowasp-yosys ?

Using from python:
yowasp-runtime==1.53
yowasp-yosys==0.44.0.0.post760

Code example:

```verilog
module main(a, result);
    input[0:8] a;
    output  [0:8] result;
    assign result = (
    (a[0:8] == 0) ? 0:
    (a[0:8] == 1) ? 1:
    (a[0:8] == 2) ? 2:
    (a[0:8] == 3) ? 3:
    (a[0:8] == 4) ? 4:
    (a[0:8] == 5) ? 5:
    (a[0:8] == 6) ? 6:
    (a[0:8] == 7) ? 7:
    (a[0:8] == 8) ? 8:
    (a[0:8] == 9) ? 9:
    (a[0:8] == 10) ? 10:
    (a[0:8] == 11) ? 11:
    (a[0:8] == 12) ? 12:
    (a[0:8] == 13) ? 13:
    (a[0:8] == 14) ? 14:
    (a[0:8] == 15) ? 15:
    (a[0:8] == 16) ? 16:
    (a[0:8] == 17) ? 17:
    (a[0:8] == 18) ? 18:
    (a[0:8] == 19) ? 19:
    (a[0:8] == 20) ? 20:
    (a[0:8] == 21) ? 21:
    (a[0:8] == 22) ? 22:
    (a[0:8] == 23) ? 23:
    (a[0:8] == 24) ? 24:
    (a[0:8] == 25) ? 25:
    (a[0:8] == 26) ? 26:
    (a[0:8] == 27) ? 27:
    (a[0:8] == 28) ? 28:
    (a[0:8] == 29) ? 29:
    (a[0:8] == 30) ? 30:
    (a[0:8] == 31) ? 31:
    (a[0:8] == 32) ? 32:
    (a[0:8] == 33) ? 33:
    (a[0:8] == 34) ? 34:
    (a[0:8] == 35) ? 35:
    (a[0:8] == 36) ? 36:
    (a[0:8] == 37) ? 37:
    (a[0:8] == 38) ? 38:
    (a[0:8] == 39) ? 39:
    (a[0:8] == 40) ? 40:
    (a[0:8] == 41) ? 41:
    (a[0:8] == 42) ? 42:
    (a[0:8] == 43) ? 43:
    (a[0:8] == 44) ? 44:
    (a[0:8] == 45) ? 45:
    (a[0:8] == 46) ? 46:
    (a[0:8] == 47) ? 47:
    (a[0:8] == 48) ? 48:
    (a[0:8] == 49) ? 49:
    (a[0:8] == 50) ? 50:
    (a[0:8] == 51) ? 51:
    (a[0:8] == 52) ? 52:
    (a[0:8] == 53) ? 53:
    (a[0:8] == 54) ? 54:
    (a[0:8] == 55) ? 55:
[... 514 lines cut ...]
    2: 0x86e118 - <unknown>!<wasm function 13577>
    3: 0xa981dd - <unknown>!<wasm function 17556>
    4: 0xa981bd - <unknown>!<wasm function 17556>
    5: 0xab6236 - <unknown>!<wasm function 17583>
    6: 0xa9fb3e - <unknown>!<wasm function 17583>
    7: 0xab7c5d - <unknown>!<wasm function 17594>
    8: 0xac47c7 - <unknown>!<wasm function 17600>
    9: 0xd6fbea - <unknown>!<wasm function 20100>
   10: 0x8c44da - <unknown>!<wasm function 14156>
   11: 0x8c3eb6 - <unknown>!<wasm function 14154>
   12: 0x8c83df - <unknown>!<wasm function 14189>
   13: 0x159f753 - <unknown>!<wasm function 26453>
   14: 0x8c8654 - <unknown>!<wasm function 14191>
   15: 0x159f3af - <unknown>!<wasm function 26452>
   16: 0x8c44da - <unknown>!<wasm function 14156>
   17: 0x8c3eb6 - <unknown>!<wasm function 14154>
   18: 0x8a3ec1 - <unknown>!<wasm function 13934>
   19: 0x889f6f - <unknown>!<wasm function 13914>
   20: 0x870f92 - <unknown>!<wasm function 13595>
   21: 0xd769 - <unknown>!<wasm function 23>

Caused by:
    0: memory fault at wasm address 0x10000000b in linear memory of size 0x9b0000
    1: wasm trap: out of bounds memory access
```

whitequark, 2024-08-22:
Unlimited: no.

Bigger; yes, I'll bump the stack size.

whitequark, 2024-08-22:
I want to point out that a 512 level deep unbalanced ternary expression is not "moderately deep", but rather "so deep that it will only appear in generated code, and even then rarely". 

In any case, I've been looking at the Yosys stack usage and under Valgrind's DRD it appears to be 24K per nesting level, or about 2.4M per 100 levels. On Linux you have about 8M of stack total and it seems that stack use is somewhat more conservative, at about 1.5M per 100 levels.

I'll increase stack size to 8M, which should bring it roughly on par with Linux. I think on macOS you only get 1M of stack, which is less than 100 levels deep. I'm not sure what happens on Windows.

whitequark, 2024-08-22:
I filed https://github.com/YosysHQ/yosys/issues/4562 to track the rather poor performance of Yosys on such expressions.

whitequark, 2024-08-22:
The next build of YoWASP Yosys on Test PyPI should have a stack size of 8M. Let me know if you need a backport of this change to the latest release branch (which would be Yosys 0.44).

closed 2024-09-03 by whitequark