YoWASP/yosys #38: Failing WebAssembly.instantiate()

qarlosalberto, 2024-09-21:
I'm using `@yowasp/yosys` version `0.43.36-dev.746` in TypeScript: 

```
const dynamicImport = eval('import("@yowasp/yosys")');
const { runYosys } = await dynamicImport;
const result1 = await runYosys(["--version"]);
```

The import works fine, but I have an error running Yosys:

```
TypeError: WebAssembly.instantiate(): Argument 0 must be a buffer source
    at init (/home/carlos/repo/vscode-terosHDL/node_modules/@yowasp/yosys/gen/bundle.js:4849:36)
```

whitequark, 2024-09-21:
Why use `eval()` there?

whitequark, 2024-09-21:
In any case, the reason it breaks is likely that `@yowasp/yosys/gen/resources-yosys.js` has this:

```
export const modules = {
    "yosys.core.wasm": new URL("./yosys.core.wasm", import.meta.url),
    "yosys.core2.wasm": new URL("./yosys.core2.wasm", import.meta.url),
    "yosys.core3.wasm": new URL("./yosys.core3.wasm", import.meta.url),
    "yosys.core4.wasm": new URL("./yosys.core4.wasm", import.meta.url),
};
```

So whatever it is you're doing with `import` probably breaks `import.meta.url`.

qarlosalberto, 2024-09-21:
Because I'm using commonjs and tsc converts imports to require. With eval I can skip that.

whitequark, 2024-09-21:
YoWASP packages are ESM-only. I think that running a bundler on `@yowasp/yosys` first might help.

qarlosalberto, 2024-09-21:
What do you mean?

whitequark, 2024-09-21:
The `new URL(..., import.meta.url)` construct is recognized by some bundlers, like esbuild, which can transform it to another construct that is supported by your environment.

qarlosalberto, 2024-09-21:
Ah okok. Thanks!!

closed 2024-09-21 by qarlosalberto