YoWASP/yosys #39: Assertion failure during step `2.23.16.6 Executing AIGER frontend`

plaes, 2025-09-03:
When attempting to run any examples from apicula, it fails with assertion:

```
pip install yowasp-yosys # (in empty venv)
git clone git@github.com:YosysHQ/apicula.git && cd apicula/examples
yowasp-yosys -D LEDS_NR=6 -D INV_BTN=0 -p "read_verilog blinky.v; synth_gowin -json blinky.json"

...
2.23.16.6. Executing AIGER frontend.
<suppressed ~1 debug messages>
ERROR: Assert `count_id(wire->name) == 0' failed in ../yosys-src/kernel/rtlil.cc:2621.

```

Version:
`Yosys 0.56 (git sha1 9c447ad9d, ccache clang++ 18.1.3 -O3 -flto -flto)`

I'm not seeing this issue with native version: `Yosys 0.56+30 (git sha1 8c71226d0050b3ff71e9bc44297480811a21bb56, ...)` installed from Fedora packages (version 0.56-1.20250810git8c71226.fc42 ).

Also, there's another issue that `yowasp-nextpnr-himbaechel-gowin` package is outdated...

whitequark, 2025-09-03:
You're hitting https://github.com/YosysHQ/yosys/issues/5274 probably; try this version: https://pypi.org/project/yowasp-yosys/0.56.0.141.post974.dev0/.

> Also, there's another issue that `yowasp-nextpnr-himbaechel-gowin` package is outdated...

See https://github.com/YosysHQ/nextpnr/issues/1546

plaes, 2025-09-03:
> You're hitting [YosysHQ/yosys#5274](https://github.com/YosysHQ/yosys/issues/5274) probably; try this version: https://pypi.org/project/yowasp-yosys/0.56.0.141.post974.dev0/.

Thanks, that one seems to work.

> > Also, there's another issue that `yowasp-nextpnr-himbaechel-gowin` package is outdated...
> 
> See [YosysHQ/nextpnr#1546](https://github.com/YosysHQ/nextpnr/issues/1546)